nRF24/RF24Ethernet #3: Won't compile in Arduino IDE 1.0.6

TMRh20, 2015-02-05:
Workaround is to use Arduino IDE 1.5.8

Per error report from blog comment.

y8, 2015-04-22:
Also can't compile `InteractiveServer_Mesh.ino` with 1.6.3:

````
Arduino: 1.6.3 (Mac OS X), Board: "Arduino Mini, ATmega328"

In file included from ~/Documents/Arduino/libraries/RF24Ethernet/RF24Ethernet.h:20:0,
                 from InteractiveServer_Mesh.ino:27:
~/Documents/Arduino/libraries/RF24Ethernet/uip-conf.h:81:34: error: 'MAX_PAYLOAD_SIZE' was not declared in this scope
 #define UIP_CONF_BUFFER_SIZE     MAX_PAYLOAD_SIZE
                                  ^
~/Documents/Arduino/libraries/RF24Ethernet/uip-conf.h:156:30: note: in expansion of macro 'UIP_CONF_BUFFER_SIZE'
   #define OUTPUT_BUFFER_SIZE UIP_CONF_BUFFER_SIZE - UIP_CONF_LLH_LEN - UIP_TCPIP_HLEN
                              ^
~/Documents/Arduino/libraries/RF24Ethernet/RF24Client.h:49:17: note: in expansion of macro 'OUTPUT_BUFFER_SIZE'
  uint8_t myData[OUTPUT_BUFFER_SIZE];
                 ^
~/Documents/Arduino/libraries/RF24Ethernet/uip-conf.h:81:34: error: 'MAX_PAYLOAD_SIZE' was not declared in this scope
 #define UIP_CONF_BUFFER_SIZE     MAX_PAYLOAD_SIZE
                                  ^
~/Documents/Arduino/libraries/RF24Ethernet/uip-conf.h:156:30: note: in expansion of macro 'UIP_CONF_BUFFER_SIZE'
   #define OUTPUT_BUFFER_SIZE UIP_CONF_BUFFER_SIZE - UIP_CONF_LLH_LEN - UIP_TCPIP_HLEN
                              ^
~/Documents/Arduino/libraries/RF24Ethernet/RF24Client.h:50:19: note: in expansion of macro 'OUTPUT_BUFFER_SIZE'
  uint8_t myDataIn[OUTPUT_BUFFER_SIZE]; 
                   ^
HTML.h: In function 'void sendPage(RF24Client&, const char*, size_t)':
~/Documents/Arduino/libraries/RF24Ethernet/uip-conf.h:81:34: error: 'MAX_PAYLOAD_SIZE' was not declared in this scope
 #define UIP_CONF_BUFFER_SIZE     MAX_PAYLOAD_SIZE
                                  ^
~/Documents/Arduino/libraries/RF24Ethernet/uip-conf.h:192:35: note: in expansion of macro 'UIP_CONF_BUFFER_SIZE'
   #define UIP_CONF_RECEIVE_WINDOW UIP_CONF_BUFFER_SIZE - UIP_CONF_LLH_LEN - UIP_TCPIP_HLEN //This is set automatically to the max allowable size
                                   ^
~/Documents/Arduino/libraries/RF24Ethernet/uip-conf.h:195:26: note: in expansion of macro 'UIP_CONF_RECEIVE_WINDOW'
 #define UIP_CONF_TCP_MSS UIP_CONF_RECEIVE_WINDOW
                          ^
~/Documents/Arduino/libraries/RF24Ethernet/utility/uipopt.h:321:21: note: in expansion of macro 'UIP_CONF_TCP_MSS'
 #define UIP_TCP_MSS UIP_CONF_TCP_MSS
                     ^
HTML.h:95:20: note: in expansion of macro 'UIP_TCP_MSS'
   size_t bufSize = UIP_TCP_MSS;
                    ^
HTML.h: In function 'void main_page(RF24Client&)':
~/Documents/Arduino/libraries/RF24Ethernet/uip-conf.h:81:34: error: 'MAX_PAYLOAD_SIZE' was not declared in this scope
 #define UIP_CONF_BUFFER_SIZE     MAX_PAYLOAD_SIZE
                                  ^
~/Documents/Arduino/libraries/RF24Ethernet/uip-conf.h:156:30: note: in expansion of macro 'UIP_CONF_BUFFER_SIZE'
   #define OUTPUT_BUFFER_SIZE UIP_CONF_BUFFER_SIZE - UIP_CONF_LLH_LEN - UIP_TCPIP_HLEN
                              ^
HTML.h:129:11: note: in expansion of macro 'OUTPUT_BUFFER_SIZE'
   char bf[OUTPUT_BUFFER_SIZE];
           ^
In file included from InteractiveServer_Mesh.ino:29:0:
HTML.h:132:15: error: 'bf' was not declared in this scope
     sprintf_P(bf,PSTR("<tr><td bgcolor=%s>\n"),lColor);
               ^
HTML.h:136:15: error: 'bf' was not declared in this scope
     sprintf_P(bf,PSTR("<tr><td> </td><td bgcolor=%s>\n"),lColor);
               ^
HTML.h:140:17: error: 'bf' was not declared in this scope
   _client.write(bf);  
                 ^
In file included from ~/Documents/Arduino/libraries/RF24Ethernet/RF24Ethernet.h:20:0,
                 from InteractiveServer_Mesh.ino:27:
HTML.h: In function 'void stats_page(RF24Client&)':
~/Documents/Arduino/libraries/RF24Ethernet/uip-conf.h:81:34: error: 'MAX_PAYLOAD_SIZE' was not declared in this scope
 #define UIP_CONF_BUFFER_SIZE     MAX_PAYLOAD_SIZE
                                  ^
~/Documents/Arduino/libraries/RF24Ethernet/utility/uipopt.h:398:21: note: in expansion of macro 'UIP_CONF_BUFFER_SIZE'
 #define UIP_BUFSIZE UIP_CONF_BUFFER_SIZE
                     ^
HTML.h:189:70: note: in expansion of macro 'UIP_BUFSIZE'
   sprintf_P(buffer, PSTR("</td><td>%u bytes</td></tr><tr><td>User "),UIP_BUFSIZE);
                                                                      ^
Error compiling.
````

TMRh20, 2015-04-22:
Likely just need to update rf24network and myb rf24mesh due to recent
changes.

y8, 2015-04-22:
Yeah, just found somewhere in your comments that "Development" branch of RF24Network is required.

Could you please, specify the requirements for RF24 libs in the README?

Thanks!

TMRh20, 2015-04-23:
Thanks for the input. It can be hard to tell what & where info is lacking, since I rarely need to reference my own documentation.

brunnels, 2015-05-17:
I believe this is associated with https://github.com/TMRh20/RF24/issues/111

TMRh20, 2015-05-18:
This issue was resolved in https://github.com/TMRh20/RF24Ethernet/commit/a92d9ff99cd030b34ea3348ad157c195452c5845, and should probably have been closed.

closed 2015-09-24 by TMRh20