nRF24/RF24Ethernet #5: Missing Definitions in RF24Ethernet_config.h

GiovanniCmp, 2015-06-14:
Hi!
I'm using Arduino UNO (Tested on IDE 1.5.8 and 1.6.3) and when I try to upload the ```Getting_Started_SimpleClient_Mesh.ino``` I get the following error:

```
In file included from C:\Users\Giovanni\Documents\Arduino\libraries\RF24Ethernet/RF24Ethernet.h:20:0,
                 from Getting_Started_SimpleClient_Mesh.ino:23:
C:\Users\Giovanni\Documents\Arduino\libraries\RF24Ethernet/uip-conf.h:81:34: error: 'MAX_PAYLOAD_SIZE' was not declared in this scope
 #define UIP_CONF_BUFFER_SIZE     MAX_PAYLOAD_SIZE
                                  ^
C:\Users\Giovanni\Documents\Arduino\libraries\RF24Ethernet/uip-conf.h:156:30: note: in expansion of macro 'UIP_CONF_BUFFER_SIZE'
   #define OUTPUT_BUFFER_SIZE UIP_CONF_BUFFER_SIZE - UIP_CONF_LLH_LEN - UIP_TCPIP_HLEN
                              ^
C:\Users\Giovanni\Documents\Arduino\libraries\RF24Ethernet/RF24Client.h:49:17: note: in expansion of macro 'OUTPUT_BUFFER_SIZE'
  uint8_t myData[OUTPUT_BUFFER_SIZE];
                 ^
C:\Users\Giovanni\Documents\Arduino\libraries\RF24Ethernet/uip-conf.h:81:34: error: 'MAX_PAYLOAD_SIZE' was not declared in this scope
 #define UIP_CONF_BUFFER_SIZE     MAX_PAYLOAD_SIZE
                                  ^
C:\Users\Giovanni\Documents\Arduino\libraries\RF24Ethernet/uip-conf.h:156:30: note: in expansion of macro 'UIP_CONF_BUFFER_SIZE'
   #define OUTPUT_BUFFER_SIZE UIP_CONF_BUFFER_SIZE - UIP_CONF_LLH_LEN - UIP_TCPIP_HLEN
                              ^
C:\Users\Giovanni\Documents\Arduino\libraries\RF24Ethernet/RF24Client.h:50:19: note: in expansion of macro 'OUTPUT_BUFFER_SIZE'
  uint8_t myDataIn[OUTPUT_BUFFER_SIZE]; 
                   ^
```
So I opened the ```RF24Ethernet_config.h``` (referenced in ```uip-conf.h```) and the is no 
```#define DISABLE_USER_PAYLOADS```, ```#define MAX_PAYLOAD_SIZE``` and other #defines.

There is something wrong!

TMRh20, 2015-06-15:
Look one issue down at #3 Or see the docs page at http://tmrh20.github.io/

GiovanniCmp, 2015-06-15:
Trying things out, I found the step-by-step guide at http://tmrh20.blogspot.com.br/2014/11/rf24ethernet-for-diy-internet-of-things.html. Basically, I needed to download the ```RF24Network (Development Fork)```, instead of the normal version.
Thanks for the help!

closed 2015-09-24 by TMRh20