nRF24/RF24Ethernet #7: compilation error on sample SLIP_InteractiveServer

catan85, 2015-10-22:
Hi, I have a compilation error on sample SLIP_InteractiveServer

SLIP_InteractiveServer:44: error: no matching function for call to 'RF24EthernetClass::RF24EthernetClass(RF24&, RF24Network&)'
SLIP_InteractiveServer.ino:44:46: note: candidates are:
In file included from SLIP_InteractiveServer.ino:34:0:
C:\Users\ancattaneo\Documents\Arduino\libraries\RF24Ethernet/RF24Ethernet.h:105:3: note: RF24EthernetClass::RF24EthernetClass()
   RF24EthernetClass();

RF24Network is development version

Bye Andrea

TMRh20, 2015-10-22:
Thanks for pointing this out. 

It looks like I forgot to update the constructors, to include RF24Mesh. If you change them to the following, it should compile for you:
```
RF24 radio(7,8);
RF24Network network(radio);
RF24Mesh mesh(radio,network);
RF24EthernetClass RF24Ethernet(radio, network,mesh);
```

Please leave this open as a reminder to correct the example.

catan85, 2015-10-23:
Now it compiles fine, thanks

closed 2021-11-06 by TMRh20