nRF24/RF24Ethernet #9: Esp8266 example

ricmail85, 2015-11-27:
Hi,

I was looking at your new example with the esp8266 wifi board. The sketch you wrote has to be uploaded on the arduino or on the esp module?
How do you connect it to the arduino?

Which library do you use with esp?

Thanks,
Bye.

TMRh20, 2015-11-27:
Its used with the [Arduino ESP8266 core](https://github.com/esp8266/Arduino) which supports [these boards](https://github.com/esp8266/Arduino/blob/master/doc/boards.md)

I use a [nodeMCU devkit board](https://commons.wikimedia.org/wiki/File:NodeMCU_DEVKIT_0.9.JPG) and tested using the git version of Arduino core.

ricmail85, 2015-12-10:
Did you connect directly the esp module with the nrf24l01+ radios? 
Or you need an arduino as a bridge?

Thanks

TMRh20, 2015-12-10:
> Did you connect directly the esp module with the nrf24l01+ radios?

Yup, there is a [thread](https://github.com/TMRh20/RF24/issues/125) about ESP & RF24 that you might find useful.

closed 2016-03-23 by TMRh20