nRF24/RF24Ethernet #11: Possible bug in RF24Client::connect(IPAddress ip, uint16_t port)

wiredman, 2015-11-30:
shouldn't the UIP_CONNECT_TIMEOUT statements be UIP_CONNECTION_TIMEOUT so it matches the uip config variable??

TMRh20, 2015-11-30:
Hopefully I don't get this backwards, but the UIP_CONNECT_TIMEOUT variable is not used at all, since it is left undefined. The behaviour of this option is to disconnect from an attempted connection, if it takes too long to establish a connection.

The UIP_CONNECTION_TIMEOUT is used, and is a bit of a hack I implemented to deal with prior client issues and stalled connections. The lib will disconnect after 30 seconds of no data received by a client by default, so this can be adjusted as required.

wiredman, 2015-12-01:
to be honest until I changed the define, I was getting random stall-outs anyways? and from my explanation really they are both for the same thing. so maybe condense it down to just the one variable?

closed 2016-03-23 by TMRh20