nRF24/RF24Ethernet #12: restore from RF24 power down

jonathanhofman81, 2015-12-30:
I'm trying to create battery powered modules running RF24Ehternet. This requires to put both the arduino and the radio in power down mode. I was wondering if this is even possible?

I already tried some things and part of it seems working. If I only put the arduino in sleep mode and keep the radio running it correctly wakes up and handles ping request I send. 

However, if I put the radio in power down (according to some examples seen in RF24Network) it seems not to wakeup based from the ping request. 

On the other hand the node can perfectly send data by it self after a wakeup of the watchdog. I call update of the mesh before, but didn't verify yet if that is required. 

Now I didn't dig to deep in the sources yet, but at first glance I think it should be possible. However, maybe dynamic mesh allocation interferes here. If you could give me some directions I try to to dig a little bit deeper. 
 
BTW: TMRh20 created a very cool set of libraries here. Great work!!!

TMRh20, 2015-12-31:
Thanks, and sleeping MCU + radio is very possible, but many of the examples need some cleaning up.

The main points:
- There is a need to call ``` radio.startListening() ``` after powering the radio up
- There is a general need to prevent the MCU from accessing the radio while powered down

<sub> Both are demonstrated in the code, which can be pasted into any RF24 sketch: </sub>
```
uint32_t powerTimer = 0;
bool powerBool = true;

void loop() {

    if(millis() - powerTimer > 5000){ //Power down/up every 5 seconds
    powerTimer=millis();  
    if(powerBool){      
      radio.powerDown();            // Power down the radio
      Serial.println("IFDown");
      delay(100);                   // Make sure Serial finishes
      enterSleep();                 // *** Call user-defined MCU sleep function or comment to only power down radio ***
    }else{
      radio.powerUp();
      radio.startListening();
      Serial.println("IFUp");
    }
    powerBool = !powerBool;
  }
  //Non radio related code goes here or above  
  if(!powerBool){return;} // Don't call any radio functions when powered down
```

Sub-points:
1. Editing RF24_config.h and uncommenting ```#define FAILURE_HANDLING``` allows the MCU/radio to recover from soft failures, so the last line of code can be removed.
2. There is no need to do anything different with RF24Mesh, unless conditions dictate it. ie: If node is offline for long periods of time, usually good to release the address first.
3. [Static nodes](http://tmrh20.github.io/RF24Mesh/classRF24Mesh.html#a6aafd9dbdd198487de392f9992218f04) can be assigned to ensure enough routing nodes are available, and can remove the need for releasing/renewing mesh addresses every time. Static routing nodes can function on devices as small as ATtiny, and can use only the RF24/RF24Network libs. Requires understanding of [RF24Network addressing](http://tmrh20.github.io/RF24Network/Addressing.html).

jonathanhofman81, 2016-01-01:
Thanks for the example! Great suggestions also. I will try this out in
combination with the examples. I did already some experiments. Not all with
success however. I will come back to this.

Best regards,

Jonathan

closed 2016-03-23 by TMRh20