nRF24/RF24Ethernet #13: RF24Ethernet violates GPL V3

ntruchsess, 2016-02-18:
I see your library contains an substancial amount of code taken from my work on UIPEthernet (https://github.com/ntruchsess/arduino_uip).
This by itself is totally fine - actually I think it's great you use my code to support totally different hardware. This is what open source is about.

But UIPEthernet is licensed as GPL V3 (http://www.gnu.org/licenses/gpl-3.0.en.html). So I may ask you to add everything that is required to comply with GPL V3 chapter 5:

a) The work must carry prominent notices stating that you modified it, and giving a relevant date.
b) The work must carry prominent notices stating that it is released under this License and any conditions added under section 7. This requirement modifies the requirement in section 4 to “keep intact all notices”.
c) You must license the entire work, as a whole, under this License to anyone who comes into possession of a copy. This License will therefore apply, along with any applicable section 7 additional terms, to the whole of the work, and all its parts, regardless of how they are packaged. This License gives no permission to license the work in any other way, but it does not invalidate such permission if you have separately received it.

TMRh20, 2016-02-18:
Yeah, there would likely not be any hint of RF24Ethernet if I had not discovered your work with ENC28J60 & uIP. I have to say it is very impressive given the inherent difficulty, and was very difficult even to follow, mostly due to uIP I think. As somebody who largely learned to program by going through code & building on the work of more advanced programmers, thank you for your contributions!  
  
I've kind of been taking a vacation from programming etc, but thought this should be handled asap., I'm happy to accept your request, (although I largely avoid licensing) but just have some questions, hoping you can maybe help with?

Point a) : There are/should be some references to your software already, specifically in the examples and I think on my blog etc, but I assume a simple reference in the documentation and/or on github would be suitable?

Point b) Same? If you have a preference, just let me know.

Point c) Does this mean that all other layers, RF24Mesh, RF24Network, RF24 would also have to carry the license, or just that the entire work as released, (in a package) needs to carry it? 

Any advice, suggestions or contributions to RF24Ethernet would be welcome as well, since it seems you must have been looking through the code.

ntruchsess, 2016-02-18:
Not a lot to do. Just add the comment found e.g. in the header of RF24Server.cpp to every file the is derived from my work on UIPEthernet. Make sure you are not accidently mixing up which source a file is originally coming from (Work e.g. by Adam Dunkels does have a different copyright-notice and licence)

Fee free to modify the very first line (that contains the filename and brief descritiption) so it suits your needs.
Add your own Copyright-notice right above mine.

like so:
------------------
 RF24Client.h - Arduino implementation of a uIP wrapper class.
 Copyright (c) 2015 <tmrh20@gmail.com>
 Copyright (c) 2013 Norbert Truchsess <norbert.truchsess@t-online.de>
All rights reserved.
 This program is free software: you can redistribute it and/or modify
.....
------------------

Regarding the 'prominent place' I suggest to add a Licence-section containing a list of all files included in RF24Ethernet that are based on third party work into README.md, specifying which licence each file is using (feel free to copy and paste from https://github.com/ntruchsess/arduino_uip/blob/master/README).

Feel free to add any Licence of your own choice to the files that are entirely your own work. If you don't want to restrict the use of your work by a 'complicated' licence you still should explicitly specify that the software is free to use, redistribute and modify (otherwise other people might decide not to use it because of not knowing your intention). I personally suggest to also use GPL V3 ;-)

TMRh20, 2016-02-21:
Thanks for the info, changes added. I try to make a point of leaving the original licensing info in, but dev was a bit messy and missed it in a number of them, so thanks for pointing this out in a helpful manner. I still need to review this and other, so please leave this issue open for the time being, and feel free to point out any mistakes.

The layers I mention are lower layers used by RF24Ethernet, so they are unrelated to the code in this lib.

ntruchsess, 2016-02-23:
perfect :-)

... leaving open as per your request...

closed 2019-05-13 by TMRh20