nRF24/RF24Ethernet #14: Port to RF22 possible? Implications?

IljaSchumacher, 2016-04-11:
Hi folks,
I need thia lib to work with the RF22 driver for 443MHz.
Is it just replacing the RF24 driver with RF22?
(as far as I know they are from the same programmer and the interface should be almost the same)
The datarate of RF22 is much lower.
Are there any implications lower rate will cause?

This lib working on a free low-Frequency LoRa would open endless possibilities.

Awesome library. Very impressed.
I know all the functionalities are already there in RF22/24/xx drivers+EthernetIP+RadioHead but puzzling it together into a working version is what I tried to arlchieve myself for ages till I just found this lib.

TMRh20, 2016-04-11:
Thanks, this library should be easily portable to other devices. I tried to keep things as general as possible at this layer, mostly just using the Arduino Ethernet API to connect uIP to the RF24Network layer. I used [SerialIP](http://playground.arduino.cc/Code/SerialIP#Reference) and  [UIPEthernet](https://github.com/ntruchsess/arduino_uip) as references.

Just connect it to an underlying network or data-link layer, which is expected to know what to do with the data. Very easy if all nodes send to a central gateway, uses ARP, or a mesh config, etc.

I think one would just need to look at the underlying configuration desired. For example, you might be able to just replace the underlying network layer with radiohead itself, with only a few modifications and #defines.

The main implications with a low data-rate situation just seems to be the need to either buffer and/or discard packets at the gateway when the radios cannot keep up, but that can be managed.

q2dg, 2016-09-06:
A really silly question: could it be possible RF24Ethernet was "agnostic" from underlying RF module model used, "à la RadioHead"? This would explode the (already great) usefulness of this fantactic library.
Thanks a lot and sorry if this suggestion is crazy.

TMRh20, 2016-09-07:
As long as the underlying driver handles the functions of OSI layer3 (Network)  or can get the data from the TCP/IP stack to its destination, it can be made to work.

**User Data --> RF24Ethernet --> uIP TCP/IP Stack --> RF24Network -->> RF24 --> RF24Network --> RF24Ethernet --> User Data**

.All the TCP/IP stack does is provide an API for establishing, maintaining and terminating connections, and then it wraps the User Data in a TCP/IP header

1. **( [TCP/IP header] [User Data] )**

  The data is then fragmented and sent over the nrf24l01 radios:
  RF24Network encapsulates the data in its own header(s), but in small chunks

2. **( [RF24Network Header] [ 24 bytes of user data] )**

3. The data is then transmitted over the network in 24-byte chunks and re-assembled at the recipient
    **( [RF24Network Header] [ TCP/IP Header + User Data] )**

4. Data is passed to the TCP/IP stack, reversing the encapsulation
   **( TCP/IP Header] [User Data] )**

 Any driver that handles steps 2 & 3 can sit underneath of RF24Ethernet, and it should be relatively easy to get it working..

q2dg, 2016-09-08:
Ooh, thanks a lot for your great explanation!

closed 2021-11-06 by TMRh20