nRF24/RF24Ethernet #15: None of the arduino example sketches connect to gateway

andreibosco, 2016-06-04:
Hello. I've spent the evening reading the docs, checking your youtube videos and reading the comments in the arduino sketches to check if I've been missing something but I just can't get the RF24Ethernet examples to connect to a RF24Gateway running on a raspberry pi.

The sketches compile and upload fine, but they don't register with the gateway (nothing shows up on 'Mesh Info') and it only outputs 'start' to the serial monitor. The interesting thing is that If I use of one the RF24Mesh examples the arduino node shows up on the gateway list.

TMRh20, 2016-06-04:
Can you please try editing the RF24Mesh.cpp file and comment out or remove the line here:
https://github.com/TMRh20/RF24Mesh/blob/master/RF24Mesh.cpp#L17

andreibosco, 2016-06-04:
Now everything works beautifully. Thanks for replying so fast.

TMRh20, 2016-06-04:
Np, thanks for reporting & testing this.

closed 2016-06-04 by TMRh20