nRF24/RF24Ethernet #16: Joining nrf module to esp wifi network

dekundu, 2016-06-27:
Hi,
I quite did not understand the purpose of RF24Ethernet library. With this can I directly communicate between a nrf module and a esp 8266.
Currently I have arduino+nrf at one side, and at the other end I have esp8266+nrf. This works :). I am using RF24Mesh library in order to achieve this. Can I just remove the nrf at the esp end, and join the esp created network (esp soft ap mode) from the arduino+nrf24l01 setup using the RF24Ethernet library?
Thanks,
Debojit

TMRh20, 2016-06-28:
These are the general models for RF24Ethernet and RF24Mesh with ESP8266:

*RF24Ethernet:*

Arduino -> RF24 -> RPi (*RF24Gateway*) -> WiFi / LAN / Internet / ESP8266 /
etc.

*RF24Mesh:*

Arduino -> RF24 -> ESP8266 -> TCP/IP -> LAN/Internet

Technically the ESP8266 could act as a LAN/Internet gateway, just like RPi,
but RF24Ethernet currently uses a separate TCP/IP stack, so this is not
implemented. Users can program the ESP to pass information between RF24 and
external networks.

cjmcx, 2019-12-21:
hi TMRh20
I have a transmiter made with pic16f689(clasic example from hoperf) , work OK
I want to transmit same buf with esp 32
two esp32 work togeder
my problem is pic transmit  (W_TX_PAYLOAD_NO_ACK, &got_time, sizeof(unsigned long)), but your library can t like that

cjmcx, 2019-12-21:
sorry for English

TMRh20, 2019-12-23:
I’m not 100% sure what you are asking, but think you just need to use a slightly different API. See https://tmrh20.github.io/RF24/classRF24.html#a23bfe6502d74bb5bbccb3a7f2ba2b5ea 

And

https://tmrh20.github.io/RF24/classRF24.html#a6253607ac2a1995af91a35cea6899c31

cjmcx, 2019-12-23:
Hi TMRh20thanks for the response so quickly my problem is that I made a remote control with pic16f690, which sends an ack-free code and then goes into sleepon the reception side also with pic16f690 listening to the code and if it is the same as the one in eeprom, execute a command (turn on a light bulb)They have been operating for about 5 yearswith esp32 we made rx and tx to work between them. but pic16f690 on the rx side, it doesn't react to esp32 tx If you think you can help me, I can send the codes, esp32 rx, tx and pic16f690 rx, txthanksbest regard

 

https://tmrh20.github.io/RF24/classRF24.html#a6253607ac2a1995af91a35cea6899c31

TMRh20, 2019-12-23:
Well it’s most likely that there is just a slight difference in the radio configuration, so yeah post your code plz.

cjmcx, 2019-12-23:
`#include <SPI.h>

#include "nRF24L01.h"
#include "RF24.h"

RF24 radio(17, 5);
//const byte rxAddr[6] = "00001";
//const uint64_t pipes[2] = { 0x1001013443LL, 0x39383736c2LL };

const uint64_t pipes[2] = { 0x3443101001LL, 0x39383736c2LL };

// pic16f690 pipe configuration
//const UINT8 RX0_Address[]={0x34,0x43,0x10,0x10,0x01};//Receive address data pipe 0
//const UINT8 RX1_Address[]={0x39,0x38,0x37,0x36,0xc2};////Receive address data pipe 1

void setup()
{
  //while (!Serial);
  Serial.begin(9600);
  
  radio.begin();
  //radio.openReadingPipe(0, rxAddr);
  radio.openReadingPipe(0, pipes[0]);
  radio.openReadingPipe(1, pipes[1]);
  radio.setChannel(0x06);
  //radio.printDetails();
  radio.startListening();
  radio.printDetails();

}

void loop()
{
  if (radio.available())
  {
    char text[32] = {0};
    radio.read(&text, sizeof(text));
    Serial.print("Received : ");
    Serial.println(text);
  } else {
    //Serial.println("No Singal");
    //delay(500);
  }
}`

cjmcx, 2019-12-23:
rx esp32

cjmcx, 2019-12-23:
tx esp32
#include  <SPI.h>

#include "nRF24L01.h"
#include "RF24.h"
int msg[1];
RF24 radio(17, 5);

//const byte rxAddr[6] = "00001";
const uint64_t pipes[2] = { 0x3443101001LL, 0x39383736c2LL };

void setup()
{
  Serial.begin(9600);
  Serial.println("Start....");
  radio.begin();
  //radio.setRetries(15, 15);
  radio.enableDynamicAck();
  //radio.openWritingPipe(rxAddr);
  radio.openWritingPipe(pipes[0]);
  radio.setChannel(0x06);
  radio.printDetails();
  radio.stopListening();
}

char text[] = "DIANA I'm Here !!";
int counter = 0;
void loop()
{
//  if (counter <10) {
//    text[0] = '0';
//    text[1] = char(counter + 97);
//    counter++;
//  } 
//  else {
//    counter = 0;
//  }
  
  Serial.print("Sending.....");
  radio.write(&text, sizeof(text),1);
  Serial.print("OK : ");
  Serial.println(text[1]);
  delay(100);
}``

cjmcx, 2019-12-23:
pic rx
`        /*********************************************************
copyright(c) 2010
Title: 				RFM70 simple example based on PIC c
Current 			version: v1.0
Function:			RFM70 demo
processor: 			PIC16F690
Clock:				8M Crystal
Author:				baiguang(ISA)
Company:			Hope microelectronic Co.,Ltd.
Contact:			+86-0755-82973805-846
E-MAIL:				rfeng@hoperf.com
Data:				2010-6-18
**********************************************************/
/*********************************************************
                        ---------------
          VDD----    |VDD         VSS|  ----GND
 	       SW_1----    |RA5         RA0|	----
sensitive_LED----    |RA4         RA1|	----VDD
          IRQ----    |RA3         RA2|  ----CE
   	  PWM OUT----    |RC5         RC0|  ----CSN
     	PWM OUT----    |RC4         RC1|	----SCK
 		    SW_2 ----    |RC3         RC2|  ----MOSI
        hall ----    |RC6         RB4|  ----MISO
     	    DS ----    |RC7         RB5|	----RX
          TX ----    |RB7         RB6|	----V_rez
			            ---------------
					       pic16F690
*********************************************************/
#include "ssrRFM70.h"
#include "built_in.h"

#define SW_1		    PORTA.F5   //RB4
#define RED_PWM 		    PORTC.F5   //RB5
#define GREEN_PWM 		    PORTC.F4
#define Sensitive_LED	  PORTA.F4   //RB6
#define V_rez           PORTB.F6
#define SW_2         	  PORTC.F3   //RC7
#define hall         	  PORTC.F6   //RC6
#define DS         	    PORTC.F7

#define SW_1_DIR 		           TRISA.F5  //=0;
#define RED_PWM_DIR			       TRISC.F5  //=0;
#define GREEN_PWM_DIR			     TRISC.F4  //=0;
#define Sensitive_LED_DIR		   TRISA.F4  //=0;
#define V_rez_DIR              TRISB.F6    // =0;
#define SW_2_DIR				       TRISC.F3  //=1;
#define hall_DIR				       TRISC.F6  //=1;
#define DS_DIR  				       TRISC.F7  //=1;

UINT8  remclose;
UINT8  invata;
UINT8 temp_buf[32];

typedef struct
{

	unsigned char reach_1s				: 1;
	unsigned char reach_5hz				: 1;
}	FlagType;

FlagType	                Flag;

void init_mcu(void);
void init_port(void);
void timer2_init(void);

void power_on_delay(void);     //   1
void delay_2s(void);           //   2
void delay_200ms(void);        //   3
[... 417 lines cut ...]
//	aaa = 1;
		}
  eeprom_write(100,5);
	while(1)
	{

//	hallin = hall;
	if(SW_1&&invata==0)
	{
	sub_program_1hz();
	delay_2s();
//	sub_program_1hz();
	}
	if(SW_1)
	{
	invata = 1;
//	GREEN_LED = 1;
	}

       Receive_Packet();

	}
}

`

cjmcx, 2019-12-23:
pic tx
`
#include "RFM70.h"
//#include <pic.h>
//#include "usart.h"
//#include "usart.c"

//#define RED_LED           PORTB.F4
#define GREEN_LED           PORTB.F6
#define SW1             porta.F5
#define SW2             porta.F4

//#define RED_LED_DIR       TRISB.F4
#define GREEN_LED_DIR       TRISB.F6
#define SW1_DIR         TRISa.F5
#define SW2_DIR         TRISa.F4

UINT8  count_50hz;
UINT8  aaa;     //temporizare actionare motor
UINT8  bbb;     //confirmare transmisie-receptie
UINT8 temp_buf[32];
  unsigned char confi ;
  unsigned char us ;
UINT8* pbuf;
typedef struct
{

	unsigned char reach_1s				: 1;
	unsigned char reach_5hz				: 1;
}	FlagType;

FlagType	                Flag;

void init_mcu(void);
void init_port(void);
void timer2_init(void);

void power_on_delay(void);
void delay_200ms(void);
void delay_50ms(void);
void delay_5ms(void);
void delay_1ms(void);
void delay_20us(void);
void sub_program_1hz(void);

void Send_Packet(UINT8 type,UINT8* pbuf,UINT8 len);
void Send_NACK_Packet(void);
void Receive_Packet(void);
void SPI_Bank1_Write_Reg(UINT8 reg, UINT8 *pBuf);
void SPI_Bank1_Read_Reg(UINT8 reg, UINT8 *pBuf);
void Carrier_Test(UINT8 b_enable); //carrier test

extern void RFM70_Initialize(void);
extern void SwitchToTxMode(void);
extern void SwitchToRxMode(void);
//const  char *tx_buf[19]="IOAN GOCAN THE BEST";
const UINT8 tx_buf[19]={0x49,0x4F,0x41,0x4E,0x20,0x47,0x4F,0x43,0x41,0x4E,0x20,0x54,0x48,0x45,0x20,0x42,0x45,0x53,0x54};
const UINT8 dx_buf[19]={0x44,0x4F,0x41,0x4E,0x41,0x20,0x4F,0x41,0x4E,0x41,0x20,0x47,0x4F,0x43,0x41,0x4E,0x20,0x31,0x31};
const UINT8 cx_buf[19]={0x43,0x52,0x49,0x53,0x54,0x49,0x41,0x4E,0x41,0x2E,0x46,0x4C,0x4F,0x52,0x49,0x4E,0x41,0x2E,0x35};
UINT8 rx_buf[MAX_PACKET_LEN];
UINT8 verif;
extern const UINT8 RX0_Address[];
extern const unsigned long Bank1_Reg0_13[];

UINT8 test_data;

void init_mcu(void)
{
//	init_comms();
	init_port();
[... 338 lines cut ...]
   {
   us = 'd';
   bbb = 1;
   }
//       Receive_Packet();
//   }
   if(bbb==0)
   {
         INTCON = 0x48;
       CE = 0;
//       IRQ = 0;
       confi = SPI_Read_Reg(CONFIG);
       SPI_Write_Reg(WRITE_REG | CONFIG, 0x00);
//       chksum = porta;
       asm{
       SLEEP;
//       NOP;
       }
   }
   SPI_Write_Reg(WRITE_REG | CONFIG, confi);
   INTCON = 0xc0;
   CE = 1;

}
}`

cjmcx, 2019-12-23:
[RFM70_receiver_code.zip](https://github.com/nRF24/RF24Ethernet/files/3995849/RFM70_receiver_code.zip)
[RFM70_transmiter_code.zip](https://github.com/nRF24/RF24Ethernet/files/3995850/RFM70_transmiter_code.zip)
register setup
I change rf_ch
I don't know if the addr is ok

void RF24::openWritingPipe(uint64_t value)
{
  ******_// Note that AVR 8-bit uC's store this LSB first, and the NRF24L01(+)
  // expects it LSB first too, so we're good._****
**
  write_register(RX_ADDR_P0, reinterpret_cast<uint8_t*>(&value), addr_width);
  write_register(TX_ADDR, reinterpret_cast<uint8_t*>(&value), addr_width);
  
  
  //const uint8_t max_payload_size = 32;
  //write_register(RX_PW_P0,rf24_min(payload_size,max_payload_size));
  write_register(RX_PW_P0,payload_size);
}

cjmcx, 2019-12-23:
`void Send_Packet(UINT8 type,UINT8* pbuf,UINT8 len)
{
UINT8 fifo_sta;
SwitchToTxMode();  //switch to tx mode

fifo_sta=SPI_Read_Reg(FIFO_STATUS);	// read register FIFO_STATUS's value
if((fifo_sta&FIFO_STATUS_TX_FULL)==0)//if not full, send data (write buff)
{
	SPI_Write_Buf(type, pbuf, len); // Writes data to buffer
bbb = 0;  //need for sleep
	delay_50ms();
}
}
that function for send from pic
`that function use to send buf
`void sub_program_3hz(void)
{
UINT8 i;
UINT8 temp_buf[32];
if(Flag.reach_1s)
{
	Flag.reach_1s = 0;
for(i=0;i<19;i++)
	{
  temp_buf[i]=cx_buf[i];
}
	Send_Packet(W_TX_PAYLOAD_NOACK_CMD,temp_buf,19);
SwitchToRxMode();  //switch to Rx mode
}
}`

cjmcx, 2019-12-23:
thanks
Merry Christmas and Happy new Year

cjmcx, 2019-12-23:
for pic ,  mplab or mikroc compiler
work with anyone

cjmcx, 2019-12-23:
[SSR.zip](https://github.com/nRF24/RF24Ethernet/files/3995913/SSR.zip)
[telecomanda 2 sw + learn + gen.zip](https://github.com/nRF24/RF24Ethernet/files/3995915/telecomanda.2.sw.%2B.learn.%2B.gen.zip)

TMRh20, 2019-12-24:
Ok, first thing to look at will be the RFM70_init.c, where the radio is
configured.

The main things I notice right away are the following:

1. Most settings appear to be the same between pic and esp32
2. Channel on pic is 23
3. Dynamic payload length is enabled on pic
4. I think you have the addressing correct, it could be backwards though,
easy to test...

So on the ESP32 side, try adding radio.enableDynamicPayloads(); and
radio.setChannel(23);

Everything else seems to match up, so that should do the trick as far as
configuration goes.

cjmcx, 2019-12-24:
all four are on Channel 6
pic rfm_init
{4,0xff},//auto retransmission dalay (4000us),auto retransmission count(15)
{5,0x06},//23 channel
{6,0x17},//air data rate-1M,out power 0dbm,setup LNA gain
esp32
radio.setChannel(0x06);
  //radio.printDetails();
  radio.startListening();
  radio.printDetails();

cjmcx, 2019-12-27:
[image]
can be diferit ?

cjmcx, 2019-12-28:
esp32
startFastWrite(buf,len,multicast);
pic16
spi_write_buf(type, pbuf, len);
does the sequence of orders sent to nrf24l01 matter, because it differs from esp32 to pic16

cjmcx, 2019-12-29:
it work !!!!

cjmcx, 2019-12-29:
but with another arduino library

closed 2021-08-03 by 2bndy5