nRF24/RF24Ethernet #23: Communication Issue between SLIP_InteractiveServer and SLIP_Gateway

RSABear, 2021-04-04:
I am using the latest code downloaded from GitHub. I have very little nRF24 experience and using the libraries to learn. I successfully covered all the examples over the last few weeks up to the RF24Ethernet SLIP_InteractiveServer and SLIP_Gateway.  I have made no code changes to the examples, except put a radio.printDetails() in the code to show the configuration. From the documentation it seems the problem is RX_ADDR_P0-1 in the SLIP_InteractiveServer code. I ask your assistance to help correct the addresses.
**RF24Ethernet SLIP_InteractiveServer:** 
[image]
**RF24Ethernet SLIP_Gateway:**
[image]

TMRh20, 2021-04-05:
The RX_ADDR_P0-1 address is fine, not sure what the problem would be. The SLIP examples never worked very well, really just a proof of concept, it’s recommended to use a raspberry Pi as the gateway with RF24gateway.

RSABear, 2021-04-05:
I don't know anybody who did not curse SLIP when we used it in the 90s to network systems. I have some RPis, will hook the demo gateway up on one. Your effort in expanding the nRF24 library code over years is highly appreciated.

RSABear, 2021-04-11:
@TMRh20 - I have the SLIP example bug fixed and working, I understand the code now much better. I will post the working code shortly.
[image]
[image]

RSABear, 2021-04-14:
I have modified the example code and stripped out everything that was not needed as below. It only works while you are sending GET requests via a browser, Postman or curl. If you leave the the InteractiveServer running for a few minutes then it dies. A quick disconnect and connect on the serial port and all runs well again. It seems that when there are no packets flowing or something it quits life. I am not sure I want to continue with the example code as the model is not what I require, I only did this because it was not working at all and I wanted to learn as much as possible. I might come back to this when time permits.
I need a very remote Nano Node to be a SLIP client opening a SLIP connection (or simpler) to the Gateway as a server, make a MqTT connection via the PubSub Lib, publish and receive a couple of messages and then disconnect. 

**RF24Ethernet Serial Gateway Example using SLIP**

```cpp
/*
 * The code was modified to work with 2 x Arduino Mega 2560 board for development purposes.
 *  
 * RF24Ethernet Serial Gateway Example using SLIP *
 * This example will allow RF24Ethernet to be used with any device capable of the SLIP protocol
 *
 * When using SLIP, there are 3 main differences to the SLIP_Gateway:
 *    1. The RF24Mesh layer is used to provide MAC/IP translation
 *    2. The Master node will always have a NodeID of 0, and the IPV4 address for the Master 
 *       in this example is defined as 10.0.3.1
 *    2. The child nodes The specified RF24Mesh nodeID must be the same as the last octet of the IP
 *        ie: IP: 10.0.3.2  NodeId must be 2
 * *    3. The RF24Ethernet library must be configured for TUN
 *       a: Open the uip-conf.h file in the RF24Ethernet directory and set the #define UIP_CONF_LLH_LEN 0
 *
 * The following commands must be run on the Linux device to enable slip
 * 1. On Linux the ports are created in the dev directory, cd /dev
 * 2. Type 'ls' , note the ttyUSB<X> devices
 * 3. Connect your Arduino
 * 4. Type 'ls' and look for a new device ttyUSB<X> where <X> is a number
 * 5. Run   sudo modprobe slip
 * 6. Run   sudo slattach -L -s 115200 -p slip /dev/ttyUSB<X> &
 * 7. Note the & at the end of the previous command. This will run slattach in the background,
 *    without it, slattach will appear to hang and CTRL+C will exit.
 * 7. Run ifconfig , note the sl<X> device
 * 8. Run   sudo ifconfig sl<X> 10.10.3.1
 * 9. Run   sudo route add -net 10.10.3.0/24 gw 10.10.3.1
 * 10. The gateway is now up and running. Active RF24Ethernet nodes should be pingable.

  - Scripts used on Linux
	sudo modprobe slip
	sudo sysctl -w net.ipv4.ip_forward=1
	sudo slattach -L -s 115200 -p slip /dev/ttyACM0 &
	sudo ifconfig sl0 10.10.3.1 dstaddr 10.10.3.2

 Note: If using an ip of 192.168.3.1 for the gateway, the commands are very similar:
        ie: sudo route add -net 192.168.3.0/24 gw 192.168.3.1

 * RF24Ethernet uses the uIP stack by Adam Dunkels <adam@sics.se>
 *
 * This example demonstrates how to configure a sensor node to act as a webserver and
 * allows a user to control a connected LED by clicking links on the webpage
 *
 * The requested URL is used as input, to determine whether to turn the LED off or on
 *
 * Debug Settings:
 * This example uses a second Serial Port - Serial1 to print the debug information to 
 * in order not to interfere with the SLIP network
 * Connect a FTDI or Foca Pro to PINs 18 & 19 as well as GND
 * The Serial Port on Linux will appear in the /dev directory as /dev/ttyUSB0 or higher
 * number if you have more serial devices connected.
 * Edit prinf.h in the RF24 directory
 * */
   
#include <SPI.h>
#include <RF24.h>
#include <RF24Network.h>
#include <RF24Mesh.h>
#include <printf.h>

/**** Configure the nRF24L01 CE and CS pins ****/
RF24 radio(7, 8);
RF24Network network(radio);
RF24Mesh mesh(radio, network);
[... 464 lines cut ...]
        digitalWrite(LED_PIN, LED_ON);
        led_state = "ON";
        Serial.println("LED ON");
      }
      if (client_Req.indexOf("/OFF") >0){
        digitalWrite(LED_PIN, LED_OFF);
        led_state = "OFF";
        Serial.println("LED OFF");
      }

      //clearing string for next read
       client_Req="";

      delay(1);           // Give the web browser time to receive the data
      client.flush();
      client.stop();      // Slose the client connection
      Serial.println("Cient disconnect");

  } // end if (client)

  // We can do other things in the loop, but be aware that the loop will
  // briefly pause while IP data is being processed.

}
```

TMRh20, 2021-04-14:
I can't seem to recreate the issue, even after leaving the devices inactive for a good while, so I would have to suggest a hardware issue, loose wire etc... something else causing it not to work after a while. I noticed it was spotty after one test, but it started working again after a minute with no changes.

A couple notes:
1. Should be calling ```Ethernet.update();``` instead of ```mesh.update();``` in the RF24_SLIP_InteractiveServer example code
2. The pubsub library seems to be a bit unreliable, suggest using the alternate mqtt library shown in 2nd mqtt example

RSABear, 2021-04-14:
Your advice is welcome, just when I think I am done. LOL! So I guess it is working after all, such a fantastic idea and solution. I would definitely not rule hardware out. I did a lot of reading and implemented all the fixes mentioned on the Forums. I bought the first two nRF24L01+ PA LNA boards plus interfaces from a local online shop, the quality is definitely not consumer grade. I have a 10 pack of each en route somewhere between China and home, then I can swap devices around to eliminate issues. Still learning, the fun and games will start when distance increases between the devices and the baud rate has to slow down, the packet size decrease and the noise level increases as the signal strength drops.

RSABear, 2021-04-24:
@TMRh20, it is not the hardware. I now have some more radios and also found the RadioHead libraries to play with. 
However, for IP over a radio network I will spare no effort. With just the Ethernet stack running, I noticed that a lot of pings ago astray so a started there.

The methodology I followed:
1. If I send a ping, I can capture the ping using wireshark.
2. Then I can dump the input buffer and compare the wireshark data with the Arduino buffer.

My findings as follows. Maybe it is suppose to be like this, but I am not sure:

1. Ping data capture on sl0 the slip interface of the Linux machine:
0000   45 00 00 54 a7 e8 40 00 40 01 78 aa 0a 0a 03 02
0010   0a 0a 03 01 00 00 1d 12 00 b8 00 01 2a 84 84 60
0020   00 00 00 00 6a 7d 0a 00 00 00 00 00 10 11 12 13
0030   14 15 16 17 18 19 1a 1b 1c 1d 1e 1f 20 21 22 23
0040   24 25 26 27 28 29 2a 2b 2c 2d 2e 2f 30 31 32 33
0050   34 35 36 37

2. Data from dump. The xxxx is first 12 bytes in the data buffer which I cannot figure out (yet). Then the icmp ping data follows, but is 12 bytes short. So I took a wild guess and added 12 bytes to the length of uip_len and get the full frame.  

-- Output from Serial Debugging
172599: MAC Received id 36951 from 00 to 05 type 149
172601: NET message 1514
172603: NET Enqueue @0 [Reading 20 bytes 0 blanks]
write_register(07,40)
172616: MAC Received id 36951 from 00 to 05 type 150
172619: NET message 2d2c
172621: NET Enqueue @0 
172623: RX 54(84) bytes
xxxx   00 00 05 00 57 90 83 00 54 00 D4 05 
0000   45 00 00 54 A7 E8 40 00 40 01 78 AA 0A 0A 03 01 
0010   0A 0A 03 02 08 00 15 12 00 B8 00 01 2A 84 84 60 
0020   00 00 00 00 6A 7D 0A 00 00 00 00 00 10 11 12 13 
0030   14 15 16 17 18 19 1A 1B 1C 1D 1E 1F 20 21 22 23 
0040   24 25 26 27 28 29 2A 2B 
*After*
0000   45 00 00 54 A7 E8 40 00 40 01 78 AA 0A 0A 03 01 
0010   0A 0A 03 02 08 00 15 12 00 B8 00 01 2A 84 84 60 
0020   00 00 00 00 6A 7D 0A 00 00 00 00 00 10 11 12 13 
0030   14 15 16 17 18 19 1A 1B 1C 1D 1E 1F 20 21 22 23 
0040   24 25 26 27 28 29 2A 2B 2C 2D 2E 2F 30 31 32 33 
0050   34 35 36 37 
write_register(00,0e)
write_register(02,3f)
172683: NET Sending id 90 from 05 to 00 type 148
172686: NET message 12 1d 00 00 01 03 0a 0a 02 03 0a 0a aa 78 01 40 00 40 e8 a7 54 00 00 45 
172694: MAC Sending to 00 via 00 on pipe 5
write_register(01,3f)
172700: NET Pipe 5 on node 00 has address cccccccce3
[Writing 32 bytes 0 blanks]

****************************

Code 1 - added a few lines of debug code in RF24Ethernet.cpp
```cpp
void RF24EthernetClass::tick()
{

    #if defined (ARDUINO_ARCH_ESP8266)
        yield();
    #endif

    if (RF24Ethernet.network.update() == EXTERNAL_DATA_TYPE) {

        if (RF24Ethernet.network.frag_ptr->message_size <= UIP_BUFSIZE) {
           uip_len = RF24Ethernet.network.frag_ptr->message_size;

           // Bernard - debug code to trace hang-up
           // 1. Serial.println(); Serial.print(millis()); Serial.print(F(" uip_len=")); Serial.println(uip_len);
           // 2. Dump the incoming data for analysis:
           printf("\n\r%lu: RX %X(%d) bytes\n\r", millis(), uip_len, uip_len); 
[... 231 lines cut ...]

  if(millis()-led_timer > led_time){                  // Pulse...
    led_timer = millis();
      if (led_control) {
        digitalWrite(DEBUG_LED_PIN, LED_ON);
        led_control = 0;
      } else {
        digitalWrite(DEBUG_LED_PIN, LED_OFF);
        led_control = 1;
    }
  }
 
  // This is the last of the differences between this and the regular Interactive Server example
  // If the master node is completely down, and unresponsive for 30 seconds, renew the address
    if(millis()-mesh_timer > mesh_time){                  // Every ss seconds, test mesh connectivity
      mesh_timer = millis();
      if( ! mesh.checkConnection() ){
        mesh.renewAddress();                              //refresh the network address
      }
    }
  
  Ethernet.update();    // Keep the network stack updated

}
```

TMRh20, 2021-04-29:
>  it is not the hardware

Are you sure? I'm unable to replicate the problem of it failing after a few minutes and the PA + LNA modules are notorious for having issues with power supplies. 
I'm not sure what to make of your results otherwise.

RSABear, 2021-04-29:
Power problems, you are correct. I have the modules connected to a Riden RD6006 for development and testing. I had to go from 10uF up to a 100uF decoupling capacitor to get the units stable, followed the advise on many posts and forums.
Sorry about my confusing post. I will report on the issue once I can produce meaningful log results between what is transmitted and received. I have spend time to create a trace and data dumps that is less engineering and more network related. The RF24Network part runs 24x7 for days.

RSABear, 2021-05-17:
When time permits I have been been coding and made progress on the issue. Just using a simple ICMP ping between the gateway and the server, embedding debug printing and reading most of the code and include files - I have a better understanding and view of what goes wrong. 90% of the time I can more or less predict when the hang-up is going to occur. 
- After start-up + 1 min. I ping from the Gateway to the Server a few times.
- Then I leave the two devices running for about 5 min.
- When I ping next, it either still works 100% or the First Fragment is received, then the Second Fragment and then the Radio will stop right there in an error condition - stop is perhaps the wrong word, it gets stuck and the code runs in a loop, endess loop.
- When the hang-up occurs the register RX_DR is always 1, and sometimes RX_P_NO goes to 0.  Now no data is received any longer of course. Going back to power, wires, antenna placement - I have tried everything, even tracing all the radio.startListening() and radio.stopListening pairs, looking for buffer over runs, the memcpy in the enqueue.... etc. 
- In the Class function RF24Network::update() I can now see the significance of radio.failureDetected = 1, as the software goes into an endless loop, running RF24Network::update() over and over from Loop(). 
- No matter what I have tried, I simply can't find a way to re-initialize the radio from scratch when the code is back on Loop()
If it was just as simple as something like:
```cpp

      if(radio.failureDetected){
        printf("loop() - failureDetected\n\r");
        radio.powerDown();
        report_failure();                    // Blink leds, send a message, etc. to indicate failure
        radio.powerUp();
        radio.begin();                       // Attempt to re-configure the radio with defaults
        mesh.setNodeID(2);
        mesh.begin();
        radio.failureDetected = 0;           // Reset the detection value
      }
```
Some more time will tell. Here is an example of my debug output:
RX = Receiving Data   
TX = Transmitting Data   
FF = First Fragment
MF = More Fragment
LF = Last Fragment
SYS = System Types

- Error:
RX<-FF   00 00 05 00 B0 1D 94 04 45 00 00 54 B4 C5 40 00 40 01 6B CD 0A 0A 03 01 0A 0A 03 02 08 00 E7 FF 
RX<-MF   00 00 05 00 B0 1D 95 03 02 F2 00 01 33 AA A2 60 00 00 00 00 76 2F 02 00 00 00 00 00 10 11 12 13 
RF24Network::update-failureDetected
STATUS		= 0xaa RX_DR=0 TX_DS=1 MAX_RT=0 RX_P_NO=5 TX_FULL=0
RF24Network::update-failureDetected
STATUS		= 0xaa RX_DR=0 TX_DS=1 MAX_RT=0 RX_P_NO=5 TX_FULL=0
RF24Network::update-failureDetected
STATUS		= 0xaa RX_DR=0 TX_DS=1 MAX_RT=0 RX_P_NO=5 TX_FULL=0
-- goes into endless loop

- Working
TX->POLL 24 09 40 00 6E 00 C2 21 
RX<-POLL 00 00 24 09 6E 00 C2 21 
TX->REQ  24 09 00 00 6E 00 C3 02 
TX->POLL 24 09 40 00 6F 00 C2 02 
TX->POLL 24 09 40 00 70 00 C2 02 
TX->POLL 24 09 40 00 71 00 C2 02 
TX->POLL 24 09 40 00 72 00 C2 02 
TX->POLL 24 09 40 00 73 00 C2 02 
TX->POLL 24 09 40 00 74 00 C2 02 
TX->POLL 24 09 40 00 75 00 C2 02 
TX->POLL 24 09 40 00 76 00 C2 02 
RX<-POLL 00 00 24 09 76 00 C2 02 
TX->REQ  24 09 00 00 76 00 C3 02 
RX<-ADDR 00 00 24 09 76 00 80 02 05 00 
TX->SYS  05 00 00 00 77 00 C6 00 05 00 
RX<-SYS  00 00 05 00 77 00 C6 00 02 00 
RX<-FF   00 00 05 00 3A 00 94 04 45 00 00 54 E1 60 40 00 40 01 3F 32 0A 0A 03 01 0A 0A 03 02 08 00 14 CE 
RX<-MF   00 00 05 00 3A 00 95 03 02 F6 00 01 9E AB A2 60 00 00 00 00 D6 5B 0A 00 00 00 00 00 10 11 12 13 
RX<-MF   00 00 05 00 3A 00 95 02 14 15 16 17 18 19 1A 1B 1C 1D 1E 1F 20 21 22 23 24 25 26 27 28 29 2A 2B 
RX<-LF   00 00 05 00 3A 00 96 83 2C 2D 2E 2F 30 31 32 33 34 35 36 37 
TX->FF   05 00 00 00 78 00 94 04 45 00 00 54 E1 60 40 00 40 01 3F 32 0A 0A 03 02 0A 0A 03 01 00 00 1C CE 
TX->MF   05 00 00 00 78 00 95 03 02 F6 00 01 9E AB A2 60 00 00 00 00 D6 5B 0A 00 00 00 00 00 10 11 12 13 
TX->MF   05 00 00 00 78 00 95 02 14 15 16 17 18 19 1A 1B 1C 1D 1E 1F 20 21 22 23 24 25 26 27 28 29 2A 2B 
TX->LF   05 00 00 00 78 00 96 83 2C 2D 2E 2F 30 31 32 33 34 35 36 37 
TX->SYS  05 00 00 00 79 00 C4 1B 02 
RX<-SYS  00 00 05 00 79 00 C4 1B 05 00 
You can clearly see the ping in and out in the debug dump.

Will post back if I get to solve the issue. I very much like this Library.

RSABear, 2021-05-17:
1hr after writing the above - I now know how to break out of the endless loop back to the main loop() - it is simple, just a bit of code to fix the library. I am going to fix this.

RX<-FF   00 00 05 00 CD 24 94 04 45 00 00 54 C0 54 40 00 40 01 60 3E 0A 0A 03 01 0A 0A 03 02 08 00 0D D2 
RX<-MF   00 00 05 00 CD 24 95 03 03 1E 00 01 29 C1 A2 60 00 00 00 00 51 1A 0B 00 00 00 00 00 10 11 12 13 
RF24Network::update-failureDetected
STATUS		= 0xc1 RX_DR=1 TX_DS=0 MAX_RT=0 RX_P_NO=0 TX_FULL=1
RF24Network::update-failureDetected
STATUS		= 0xc1 RX_DR=1 TX_DS=0 MAX_RT=0 RX_P_NO=0 TX_FULL=1
loop() - failureDetected
TX->POLL 24 09 40 00 3B 3F C2 C6 
RF24Network::update-failureDetected
STATUS		= 0xc1 RX_DR=1 TX_DS=0 MAX_RT=0 RX_P_NO=0 TX_FULL=1
TX->POLL 24 09 40 00 3C 3F C2 C6 
RF24Network::update-failureDetected