nRF24/RF24Ethernet #24: Endianess issue on STM32 nodes

microtronics, 2021-07-10:
I opened an issue over at RF24Gateway ([link](https://github.com/nRF24/RF24Gateway/issues/19)) that belongs here.
Using RF24Ethernet on STM32duino the bytes if the IP adresses are swapped pairwise. I thought about this a bit and have a solution to propose:
In RF24Ethernet the following code converts the IP into the uip style:
```
 #define uip_ip_addr(addr, ip) do { \
                      ((u16_t *)(addr))[0] = HTONS(((ip[0]) << 8) | (ip[1])); \
                      ((u16_t *)(addr))[1] = HTONS(((ip[2]) << 8) | (ip[3])); \
                   } while(0)
```
The HTONS macro is used to convert endianess, but I believe it is actually not necessary to do so: uIP expects the IP as it shows up in an IP header. This order is always the same and can always be constructed the same way from IPadress, since this class is based on a uint8_t[4] array, so the order is fixed. Endianess shouldn't play any roll there.

The proposed solution would be:
```
#define uip_ip_addr(addr, ip) do { \
                     memcpy(addr, &ip[0], 4); \
                  } while(0)
```
This works on both the Arduino nano and the STM32 for me. Now, some might argue that memcpy is too blown up, but it actually compiles the arduino code to a smaller size than before. The compiler might even further optimize this, if appropriate

2bndy5, 2021-07-10:
is the `do ... while(0)` needed? I think we could get away with 
```c
#define MACRO_NAME(arg1, arg2) ({ func(arg1, arg2); })
```

but I'm not well versed in C only syntax. I doubt this would help/harm any performance though.

microtronics, 2021-07-10:
The reasoning is simply to encapsulate the macro so it won't throw errors in nested situations:
http://www.bruceblinn.com/linuxinfo/DoWhile.html
However, the way it is currently used it won't make any difference if it is left out.

2bndy5, 2021-07-11:
@TMRh20 there seems to be quite a few updates to the [UIPEthernet lib](https://github.com/UIPEthernet/UIPEthernet/releases). Actually, it looks like the lib got some fresh blood back in 2017 including some support for more boards (like STM32F series). However, with the fresh blood came a new major version, so converting the RF24Ethernet code to use the UIPEthernet v2.0.x may take some meticulous work (there's a lot of copied code). It would be better to have the UIPEthernet lib listed as a dependency (PIO & Arduino IDE) for RF24Ethernet to avoid future problems like this issue.

@microtronics Can you submit a PR to resolve this (for now)?

I should put together an Arduino CI workflow for this lib...

2bndy5, 2021-07-11:
@microtronics  what exact stm32 board are you using?

microtronics, 2021-07-11:
@2bndy5 I used a STM32L053R8 nucleo board for these tests. The PR is open. I agree that including a library would be better, but I guess that would require a major rewrite...

2bndy5, 2021-07-11:
> I guess that would require a major rewrite...

Yeah, I've been looking into it, and it looks like some of the uip lib is altered for this nRF24L01 transceiver as the network adapter. So, maybe not a good idea on my part.

> I used a STM32L053R8 nucleo board for these tests

I asked because no stm32 boards I tried could compile in PlatformIO using Arduino framework (when putting together the CI workflow). I will try add that board and see how it goes.

microtronics, 2021-07-11:
Yep, the problem is: RF24Ethernet uses a utility class "timer.h", but there is also a "timer.h" in the STM32 framwork. In platformio this clashes, but Arduino did not have this issue. Simple fix ist to rename timer.h in the library, though I think this should also be resolved by PIO. Yeah, same names for headers and sources are evil, but you can never guarantee it in growing projects...

2bndy5, 2021-07-11:
Thanks for verifying my experience & providing the _real_ solution (compiler output isn't much help there). I'll keep the `ststm32` platform out of the library.json for PIO packaging.

commit b2bae576 (2021-07-11): Merge pull request #25 from microtronics/stm32fix

Use memcpy instead of byte packing. Fixes #24
closed 2021-07-11 by TMRh20