nRF24/RF24Ethernet #28: Conflicting header inclusion timer

h-sh-h, 2022-06-11:
Hi, 
Thanks for the awesome library.

Currently compiling the **RF24Ethernet** for **BluePill** in **PlatformIO** cause **conflicting inclusion**.
Both stm32duino & RF24Ethernet have **timer.h**. Since Arduino IDE & PlatformIO have different inclusion order, Compiling the library in PlatformIO project cause error. 
Already the only known workaround is to fork the library, fix the file names and use the forked library in PlatformIO project.

Thanks

TMRh20, 2022-06-11:
Thanks for reporting this. We will get onto fixing it right away. We should be due for a release soon too, so this should get pushed out relatively quickly

TMRh20, 2022-06-11:
@h-sh-h I added some changes to the [Fix #28 branch](https://github.com/nRF24/RF24Ethernet/tree/Fix%2328) if you could verify it works that would be great.

2bndy5, 2022-06-11:
I believe this problem also hindered the teensy boards on PlatformIO as well.

PR #29 (2022-06-15): Fix for #28
Rename timer.h and timer.c

Closes #28