nRF24/RF24Ethernet #43: Main URL in client examples is no longer valid

TMRh20, 2024-03-20:
Reminder to update client examples with a new URL since http://fiikus.net/asciiart/pizza.txt is no longer valid. 
Hard to find links to good ascii art that don't use SSL...

2bndy5, 2024-03-20:
Yeah, SSL is pretty much standard nowadays. Firefox actually tries to prevent me from visiting sites that don't use https.

What about ftp?

TMRh20, 2024-03-20:
Using FTP might require some slight changes to the code, but should work as a good alternative.

TMRh20, 2024-03-21:
Found an unsecure site with a bunch of ASCII art etc:  http://artscene.textfiles.com/asciiart/texthistory.txt

I think we can use this.

TMRh20, 2024-03-21:
Now it seems the secondary URL is down as well lol. But I also found http://mewbies.com/geek_fun_files/scarecrow_ascii_art_archive/19941205_ASCII%20Art%20Resources.htm

2bndy5, 2024-03-21:
On my android, firefox automatically replaces the URL's http with https. Obviously, https does not work.

TMRh20, 2024-03-23:
Interesting, that's probably best-practice nowadays. There's probably a setting to allow non-secure sites you would think. 
Looks like the backup URL is back up now too [here](http://109.120.203.163/web/blyad.club/library/litrature/Salvatore,%20R.A/Salvatore,%20R.A%20-%20Icewind%20Dale%20Trilogy%201%20-%20Crystal%20Shard,%20The.txt)

closed 2024-04-09 by TMRh20