nRF24/RF24Ethernet #50: TCP/IP via two ESP32 devices

epsi1on, 2024-11-28:
Hi,

Please consider this topology:
[image]
Think of device 1 want to ping device 2 with Linux `ping` command. I would be thankful if you give me some guidance.
I would like to know if it is possible to make NRF24L01, transparent for communication so that two devices can directly communicate via TCP or UDP. if not, how can i get a tcp or udp stream between two ESP32 Wroom devices? I'm not limited to ESP32 so can use other micro controllers as well.
Thanks for great work

2bndy5, 2024-11-28:
I believe [RF24Gateway](https://github.com/nRF24/RF24Gateway) is the sister library to RF24Ethernet but for Linux. Unfortunately, we don't have a python binding on Linux for RF24Gateway.

epsi1on, 2024-11-29:
Thanks for reply.
I actually mean something like the [nerfnet](https://github.com/aarossig/nrfnet) which two raspberry pi clients can communicate via TCP/IP together via NRF medium. each one have an IP and could ping each other with ping command in linux.
I mean such level of integrity into linux device, not a simply python binding. It is actually a little hard for me to understand wether RF24Gateway or RF24Ethernet could be used for that purpose.

TMRh20, 2024-11-29:
Yes, that is basically what RF24Gateway does, it uses the nRF24 radio as a standard NIC.

This allows users to utilize virtually any standard protocol that works over TCP/IP or UDP, without needing to do any specific coding etc. Just set up the devices with RF24Gateway and it works like a standard network card.

epsi1on, 2024-11-30:
Kindly thank you

closed 2024-11-30 by epsi1on